STMicroelectronics/STM32CubeF3 #1: HRTIM_TIM_TypeDef is undefined

salkinium, 2019-06-04:
**Describe the set-up**
 * arm-none-eabi-gcc
 * `#include <stm32f3xx.h>`

**Describe the bug**

`HRTIM_TIM_TypeDef` is undefined. After cross referencing this with the RM you probably meant to use `HRTIM_Timerx_TypeDef`.

**How To Reproduce**

Trying to access `HRTIM1_TIMA` in any code will fail, since `HRTIM_TIM_TypeDef` is undefined:

```c
#include <stm32f3xx.h>
HRTIM1_TIMA->TIMxCR = 0;
```

**Additional context**

```diff
-#define HRTIM1_TIMA         ((HRTIM_TIM_TypeDef *) HRTIM1_TIMA_BASE)
-#define HRTIM1_TIMB         ((HRTIM_TIM_TypeDef *) HRTIM1_TIMB_BASE)
-#define HRTIM1_TIMC         ((HRTIM_TIM_TypeDef *) HRTIM1_TIMC_BASE)
-#define HRTIM1_TIMD         ((HRTIM_TIM_TypeDef *) HRTIM1_TIMD_BASE)
-#define HRTIM1_TIME         ((HRTIM_TIM_TypeDef *) HRTIM1_TIME_BASE)
+#define HRTIM1_TIMA         ((HRTIM_Timerx_TypeDef *) HRTIM1_TIMA_BASE)
+#define HRTIM1_TIMB         ((HRTIM_Timerx_TypeDef *) HRTIM1_TIMB_BASE)
+#define HRTIM1_TIMC         ((HRTIM_Timerx_TypeDef *) HRTIM1_TIMC_BASE)
+#define HRTIM1_TIMD         ((HRTIM_Timerx_TypeDef *) HRTIM1_TIMD_BASE)
+#define HRTIM1_TIME         ((HRTIM_Timerx_TypeDef *) HRTIM1_TIME_BASE)
```

[Full patch](https://github.com/modm-io/cmsis-header-stm32/blob/master/patches/f3.patch)
Related to https://github.com/STMicroelectronics/STM32CubeH7/issues/2.

CCASTM, 2019-06-05:
Hello Niklas,
Thank you for your finding. This has been communicated to the development team and we will come back to you (a very first analysis confirms the problem, as mentioned the STM32H7 series)
Regards,
Christophe

FVISTM, 2019-06-05:
Hello,
New internal Buzilla tracker created (Ticket 67768 - [CMSIS Device] Wrong definition of HRTIM1_TIMx constants)

ALABSTM, 2019-06-10:
ST Internal Reference: 67770

ALABSTM, 2019-09-20:
Hi Salkinium,

This issue has been integrated to STM32CubeF3 v1.11.0 which will be available soon. This issue could be closed in this case.

With regards,

closed 2019-09-26 by ALABSTM