STMicroelectronics/STM32CubeF3 #2: CDC does not transmit packets which length is an exact multiple of 64 bytes

JulienTitan, 2019-08-14:
The issue was already reported in the communinty forum: https://community.st.com/s/question/0D50X0000BCMIR8SQP/bug-cdc-does-not-transmit-packets-which-length-is-an-exact-multiple-of-64-bytes

**Describe the set-up**
Any board implementing a Virtual COM Port using the USB CDC class

**Describe the bug**
When using USBD_CDC_SetTxBuffer followed by USBD_CDC_TransmitPacket with a packet of 64, 128, 192, etc. bytes, the data are not sent unless they are followed by another call to those functions with data length different than multiple of 64 bytes.

**How To Reproduce**
Send a packet of bytes which length is a multiple of 64 bytes using USBD_CDC_SetTxBuffer/USBD_CDC_TransmitPacket functions.

**Additional context**
A fix in the HAL library in explained in the thread below:

https://community.st.com/s/question/0D50X00009XkXmgSAF/usb-cdc-breaks-on-heavy-load

 

The fix is in PCD_EP_ISR_Handler function of the stm32f3xx_hal_pcd.c file. In one of the last lines of the function:

    if (ep->xfer_len == 0U)

To be replaced by:

    if ((ep->xfer_len == 0U) && (ep->xfer_count < ep->maxpacket))

The last lines of the function code would look like:

            /*multi-packet on the NON control IN endpoint*/
            ep->xfer_count = PCD_GET_EP_TX_CNT(hpcd->Instance, ep->num);
            ep->xfer_buff+=ep->xfer_count;
           
            /* Zero Length Packet? */
            if ((ep->xfer_len == 0U) && (ep->xfer_count < ep->maxpacket))
            {
              /* TX COMPLETE */
              HAL_PCD_DataInStageCallback(hpcd, ep->num);
            }
            else
            {
              HAL_PCD_EP_Transmit(hpcd, ep->num, ep->xfer_buff, ep->xfer_len);
            }

CCASTM, 2019-08-19:
Hello Julian,
Thank you to report this Issue with detailed explanation and improvement proposal.

I will check with our colleagues from the development team as they could have been informed of your report in the ST Community and we will keep you informed of the analysis

Regards
Christophe

bouattay, 2019-08-19:
Hello Julian,
Required ZLP to complete multi max packet size transfers is a known issue already fixed in our USB MW device lib. This should be visible to our customers in the next F3 release.

Regards,
Aymen

CCASTM, 2019-08-20:
Hello Julian,
Just to confirm that the correction proposed is working fine.
The official MW version update is different but equivalent, it is already available in other STM32 series (F4, F7, L4 ...) and will be available in the next STM32F3 FW Package  publication at then end of 2019. 
If you have no other question, we can close this Issue.
Regards
Christophe

JulienTitan, 2019-08-20:
Hell Christophe and Aymen,
Thank you for all the replies. 
I don't have any further question and the release date makes sense to me. I agree we can close the issue.
Kind reagrds,
Julien

closed 2019-08-20 by JulienTitan

ALABSTM, 2019-11-29:
ST Internal Reference: 14144