STMicroelectronics/STM32CubeF3 #3: Comment bug stm32f3xx_ll_opamp.h NONINVERT macros

aquaid3, 2019-11-09:
In the low-level opamp driver, PB11 and PA4 for OPAMP4 are swapped in the comments after the last two non-inverting macro definitions.

stm32f3xx_ll_opamp.h:

-#define LL_OPAMP_INPUT_NONINVERT_IO2      (OPAMP_CSR_VPSEL_1)     /*!< OPAMP non inverting input connected to GPIO pin (pin PA3 for OPAMP1, pin PB0  for OPAMP2, pin PA1  for OPAMP3, pin PB11 for OPAMP4) */
-#define LL_OPAMP_INPUT_NONINVERT_IO3      (OPAMP_CSR_VPSEL_0)     /*!< OPAMP non inverting input connected to GPIO pin (pin PA5 for OPAMP1, pin PB14 for OPAMP2, pin PA5  for OPAMP3, pin PA4  for OPAMP4) */
+#define LL_OPAMP_INPUT_NONINVERT_IO2      (OPAMP_CSR_VPSEL_1)     /*!< OPAMP non inverting input connected to GPIO pin (pin PA3 for OPAMP1, pin PB0  for OPAMP2, pin PA1  for OPAMP3, pin PA4 for OPAMP4) */
+#define LL_OPAMP_INPUT_NONINVERT_IO3      (OPAMP_CSR_VPSEL_0)     /*!< OPAMP non inverting input connected to GPIO pin (pin PA5 for OPAMP1, pin PB14 for OPAMP2, pin PA5  for OPAMP3, pin PB11  for OPAMP4) */

ALABSTM, 2019-11-11:
Hi Aquaid3,

Thank you for this message. It will be immediately forwarded to the development team in charge of the OPAMP driver. I will be back to you as soon as they confirm the issue. Thank you for your patience and thank you for your message once again.

With regards,

PS: I took the initiative to add some formatting tags to your message. I hope you are okay with it.

ALABSTM, 2019-11-11:
Hi Aquaid3,

The issue has been confirmed. The comments will be updated. The correction will be available in a future release. Thank you once more.

With regards,

ALABSTM, 2019-11-11:
ST Internal Reference: 75493

closed 2020-12-02 by ALABSTM