STMicroelectronics/STM32CubeF3 #4: assert_failed in tpl_main_c.ftl, signature mismatch

crackwitz, 2020-01-21:
**Describe the set-up**
 * STM32F303CB on custom board
 * freshest CubeMX, freshest CubeF3, last release of TrueStudio

**Describe the bug**
Type mismatch due to inconsistent templates of CubeMX.

CubeMX/db/templates/tpl_main_c.ftl near line 1126 contains a special case for "stm32f0", "stm32f3", "stm32l4" to use `void assert_failed(char *file, uint32_t line)` instead of `uint8_t*` like everywhere else.

this conflicts with what you generate from `stm32f3xx_hal_conf_h.ftl`: `void assert_failed(uint8_t* file, uint32_t line);`

**How To Reproduce**
1. Generate project for STM32F303CB

**How To Fix**
1. Remove the `stm32f3` from the special case in `tpl_main_c.ftl:1126` that picks `char*` signature for `assert_failed`, so `uint8_t*` is used consistently

**Additional context**
Some grep:
```
stm32f3xx_hal_conf_h.ftl:  * @param  expr If expr is false, it calls assert_failed function
stm32f3xx_hal_conf_h.ftl:  #define assert_param(expr) ((expr) ? (void)0U : assert_failed((uint8_t *)__FILE__, __LINE__))
stm32f3xx_hal_conf_h.ftl:  void assert_failed(uint8_t* file, uint32_t line);

tpl_main_c.ftl:void assert_failed(char *file, uint32_t line)
tpl_main_c.ftl:void assert_failed(uint8_t *file, uint32_t line)
```

crackwitz, 2020-01-21:
might I recommend just switching everything to the same signature... the `char*` special case lists f0,f3,l4, but corresponding files that use `char*` exist for f2,l4 but not f0 and not f3

ALABSTM, 2020-02-05:
Hi Christoph,

First, allow me to thank you for your contribution. However, aspects related to Cube MX tool are not handled at GitHub level. They are rather treated at **ST Community** level, precisely within the [Cube MX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated section.

There, you may find posts that already raised the issue you are pointing out here. Otherwise, you can create a new post to submit your issue.

Please, allow me to close this issue. Thank you for your comprehension.

With regards,

**PS:** For you information, the prototype to consider is the one using the `uint8_t *` type, **i.e.** `void assert_failed(uint8_t* file, uint32_t line);`. The one using the `char *` type is a deprecated one.

closed 2020-02-05 by ALABSTM