STMicroelectronics/STM32CubeF3 #6: HAL_TIM_DMABurst_WriteStart does work after  V1.11.2

tiantian1645, 2020-12-28:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * STM32F302C8T6
 * STN32CubeIDE v1.5.1 

**Describe the bug**
After update this lib from   V1.11.1 to V1.11.2, the HAL_TIM_DMABurst_WriteStart function does not work as noraml.

**How To Reproduce**
1. use V1.11.1 and conf Timer and DMA as this:
```
static void MX_TIM16_Init(void)
{

    /* USER CODE BEGIN TIM16_Init 0 */

    /* USER CODE END TIM16_Init 0 */

    TIM_OC_InitTypeDef sConfigOC = {0};
    TIM_BreakDeadTimeConfigTypeDef sBreakDeadTimeConfig = {0};

    /* USER CODE BEGIN TIM16_Init 1 */

    /* USER CODE END TIM16_Init 1 */
    htim16.Instance = TIM16;
    htim16.Init.Prescaler = MOTOR_PSC;
    htim16.Init.CounterMode = TIM_COUNTERMODE_UP;
    htim16.Init.Period = MOTOR_ARR;
    htim16.Init.ClockDivision = TIM_CLOCKDIVISION_DIV1;
    htim16.Init.RepetitionCounter = MOTOR_PCR;
    htim16.Init.AutoReloadPreload = TIM_AUTORELOAD_PRELOAD_DISABLE;
    if (HAL_TIM_Base_Init(&htim16) != HAL_OK) {
        Error_Handler();
    }
    if (HAL_TIM_PWM_Init(&htim16) != HAL_OK) {
        Error_Handler();
    }
    sConfigOC.OCMode = TIM_OCMODE_PWM1;
    sConfigOC.Pulse = MOTOR_PUL;
    sConfigOC.OCPolarity = TIM_OCPOLARITY_HIGH;
    sConfigOC.OCNPolarity = TIM_OCNPOLARITY_HIGH;
    sConfigOC.OCFastMode = TIM_OCFAST_DISABLE;
    sConfigOC.OCIdleState = TIM_OCIDLESTATE_RESET;
    sConfigOC.OCNIdleState = TIM_OCNIDLESTATE_RESET;
    if (HAL_TIM_PWM_ConfigChannel(&htim16, &sConfigOC, TIM_CHANNEL_1) != HAL_OK) {
        Error_Handler();
    }
    sBreakDeadTimeConfig.OffStateRunMode = TIM_OSSR_DISABLE;
    sBreakDeadTimeConfig.OffStateIDLEMode = TIM_OSSI_DISABLE;
    sBreakDeadTimeConfig.LockLevel = TIM_LOCKLEVEL_OFF;
    sBreakDeadTimeConfig.DeadTime = 0;
    sBreakDeadTimeConfig.BreakState = TIM_BREAK_DISABLE;
    sBreakDeadTimeConfig.BreakPolarity = TIM_BREAKPOLARITY_HIGH;
    sBreakDeadTimeConfig.BreakFilter = 0;
    sBreakDeadTimeConfig.AutomaticOutput = TIM_AUTOMATICOUTPUT_DISABLE;
    if (HAL_TIMEx_ConfigBreakDeadTime(&htim16, &sBreakDeadTimeConfig) != HAL_OK) {
        Error_Handler();
    }
    /* USER CODE BEGIN TIM16_Init 2 */

    /* USER CODE END TIM16_Init 2 */
    HAL_TIM_MspPostInit(&htim16);
}

/**
 * Enable DMA controller clock
 */
static void MX_DMA_Init(void)
{

    /* DMA controller clock enable */
    __HAL_RCC_DMA1_CLK_ENABLE();

    /* DMA interrupt init */
    /* DMA1_Channel3_IRQn interrupt configuration */
    HAL_NVIC_SetPriority(DMA1_Channel3_IRQn, 5, 0);
    HAL_NVIC_EnableIRQ(DMA1_Channel3_IRQn);
}
```
 
2. start Timer PWM and call HAL_TIM_DMABurst_WriteStart 
```
HAL_TIM_PWM_Start(&htim16, TIM_CHANNEL_1);
HAL_TIM_DMABurst_WriteStart(&htim16, TIM_DMABASE_ARR, TIM_DMA_UPDATE, (uint32_t *)gMotor_SRC_Buffer, TIM_DMABURSTLENGTH_3TRANSFERS);
```

3. in HAL_TIM_PeriodElapsedCallback call HAL_TIM_DMABurst_WriteStart again
```
callback_cnt++;
if (callback_cnt > 10000) {
    HAL_TIM_DMABurst_WriteStop(&htim16, TIM_DMA_UPDATE);
    HAL_TIM_PWM_Stop(&htim16, TIM_CHANNEL_1);
} else {
    HAL_TIM_DMABurst_WriteStart(&htim16, TIM_DMABASE_ARR, TIM_DMA_UPDATE, (uint32_t *)gMotor_SRC_Buffer, TIM_DMABURSTLENGTH_3TRANSFERS);
}
```

3. switch to V1.11.2 and HAL_TIM_PeriodElapsedCallback will only Enetr 3 times

tiantian1645, 2020-12-28:
`htim->DMABurstState = HAL_DMA_BURST_STATE_READY;` only  call in function like *Init and *Stop 
When enter `HAL_TIM_PeriodElapsedCallback` the `htim->DMABurstState` is `HAL_DMA_BURST_STATE_BUSY`

ASELSTM, 2021-01-05:
Hi @tiantian1645,

Thank you for this report. For a better understanding of the issue and in order to reproduce the same behavior, would you please share the whole IOC file if it is possible, so that we can use it to reproduce the issue. 

Regarding the unknown word **"trom"** you have found within the stm32f3xx_hal_tim.h driver, this is a typo and it should be replaced by **"from"**. An internal bug tracker will be raised to fix this. Thank you again for bringing this issue to our attention. 

With regards,

tiantian1645, 2021-01-06:
Hello:
   Here is my ioc file https://pastebin.ubuntu.com/p/ZpDsp8ZZpz/
   I use TIM16 CH1 as PWM output.
   You can see another example in your _STM_ **TIM_DMABurst**.  
   Or you can refer AN4776 _General-purpose timer cookbook for STM32 microcontrollers_  - **Section 5 Arbitrary waveform generation using timer DMA-burst feature**.
    Please test before release.
Thank you

ASELSTM, 2021-01-07:
Hi @tiantian1645,

A new **DMABurstState** field has been added in the global structure definition of TIM_HandleTypeDef in order to manage DMA Burst operation state. Checks within HAL_TIM_DMABurst_MultiWriteStart and HAL_TIM_DMABurst_MultiWriteStop functions are now done on DMABurstState and no longer on the TIM operation state, htim->state. TIM operation state and DMA Burst operation state are now managed separately.

Actually, a `HAL_TIM_DMABurst_WriteStop `is required to enable once again a DMA Burst start write operation. Then, to fix the issue `HAL_TIM_PeriodElapsedCallback` should be updated as following : 

```diff
callback_cnt++;
if (callback_cnt > 10000) {
    HAL_TIM_DMABurst_WriteStop(&htim16, TIM_DMA_UPDATE);
    HAL_TIM_PWM_Stop(&htim16, TIM_CHANNEL_1);
} else {
+   HAL_TIM_DMABurst_WriteStop(&htim16, TIM_DMA_UPDATE);
    HAL_TIM_DMABurst_WriteStart(&htim16, TIM_DMABASE_ARR, TIM_DMA_UPDATE, (uint32_t *)gMotor_SRC_Buffer, TIM_DMABURSTLENGTH_3TRANSFERS);
}
```

Hope this will help you. 

With regards,

tiantian1645, 2021-01-09:
Hello: 
   What the hell??? Use stop timer code in call back hit myself?
   Why do not I direct call `htim->DMABurstState = HAL_DMA_BURST_STATE_READY;`?
   
Than you.

ASELSTM, 2021-01-15:
Hi @tiantian1645,

Indeed, it is also possible to make a direct call `htim->DMABurstState = HAL_DMA_BURST_STATE_READY;`  but it is not recommended. 

With regards,

ASELSTM, 2021-08-16:
Hi @tiantian1645,

The typos you reported has been fixed in the frame of latest version of the STM32CubeF3 published on GitHub.

Please allow me then to close this thread and thank you again for having reported.

With regards,

closed 2021-08-16 by ASELSTM