STMicroelectronics/STM32CubeF3 #10: Missing (PCD_HandleTypeDef *) casts cause build errors in C++ compilers when USB is used

pawelszramowski, 2021-02-17:
Since `void*` is not implicitly convertible to other pointers, missing `(PCD_HandleTypeDef *)` casts cause build errors in C++ compiler (I used IAR EW for Arm 8.50.1) when USB is used. This affects the following files:
* `Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c` included in firmware packages downloaded by STM32CubeMX and copied to or referenced by user projects. Replace `PCD_HandleTypeDef *hpcd = pdev->pData;` with `PCD_HandleTypeDef *hpcd = (PCD_HandleTypeDef *)pdev->pData;` to fix it.
* `usbd_conf.c` generated from `db/templates/usbdconf*_c.ftl` templates in STM32CubeMX installation directory. Replace `(HAL_PCD_.*\()(pdev->pData)` with `$1\(PCD_HandleTypeDef \*\)$2` using regular expressions to fix it in `HAL_PCD_*` function calls.

ALABSTM, 2021-04-20:
Hi @pawelszramowski,

The issue you reported has been reproduced using the C++ compiler of EWARM **v8.32.4** (an even older version than the one you are using). Your report will be forwarded to our development teams. I will get back to you as soon as I have their feedback.

With regards,

ALABSTM, 2021-04-20:
Hi @pawelszramowski,

Our development teams confirmed the cast is missing. This will be logged into our internal database and a fix will be made available in the frame of a future release.

With regards,