STMicroelectronics/STM32CubeF3 #12: unused parameter warnings

T5100, 2021-03-05:
there are some unused parameter warnings in the Driver code.

For example:
in Drivers/STM32F3xx_HAL_Driver/Inc/stm32f3xx_ll_adc.h the function LL_ADC_DMA_GetRegAddr has a unused variable variable, if ADC_MULTIMODE_SUPPORT is undefined.

#if defined(ADC_MULTIMODE_SUPPORT)
....
#else
__STATIC_INLINE uint32_t LL_ADC_DMA_GetRegAddr(ADC_TypeDef *ADCx, uint32_t **Register**)
{
  /* Retrieve address of register DR */
  return (uint32_t)&(ADCx->DR);
}
#endif

RKOUSTM, 2021-03-08:
Hi @T5100,
 
Thank you for contribution. It has been transferred to our development teams. We will be back to you as soon as we get an answer. 

In order to accept and analyze [this](https://github.com/STMicroelectronics/STM32CubeF3/pull/11) pull-request, please sign  the <a href='https://cla.st.com'>Contribution License Agreement</a>. 

Thank you for you again for your contribution.

With regards

RKOUSTM, 2021-03-08:
Hi @T5100,

Thank you for your contribution. The issue you pointed out has been confirmed, a fix will be made available in future release.

Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-03-08:
ST Internal Reference: 102614

RKOUSTM, 2021-08-16:
Hi @T5100,

The fix you requested has been implemented and is now available in the frame of the latest STM32CubeF3 FW package V1.11.3 release.

This issue can be closed now. Thank you again for your contribution.

With regards,

closed 2021-08-16 by RKOUSTM