STMicroelectronics/STM32CubeF3 #13: missing callback call in I2SEx_TxRxDMACplt() if DMA is in circular mode

lukasnee, 2021-04-05:
Problem location: https://github.com/STMicroelectronics/STM32CubeF3/blob/master/Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_i2s_ex.c
function I2SEx_TxRxDMACplt(...)

I think this callback logic
```
 /* Call user TxRx complete callback */
#if (USE_HAL_I2S_REGISTER_CALLBACKS == 1U)
        hi2s->TxRxCpltCallback(hi2s);
#else
        HAL_I2SEx_TxRxCpltCallback(hi2s);
#endif /* USE_HAL_I2S_REGISTER_CALLBACKS */
```
is missing outside the scope
```  
/* if DMA is configured in DMA_NORMAL mode */
  if (hdma->Init.Mode == DMA_NORMAL) {

 /* ... */

}
```
I came up on this problem when tried to set up I2S full duplex DMA in circular mode. I found that 
this callback was being called periodically.
`void HAL_I2SEx_TxRxHalfCpltCallback(I2S_HandleTypeDef *hi2s)`
but not this:
`void HAL_I2SEx_TxRxCpltCallback(I2S_HandleTypeDef *hi2s)`

I copied the callback logic (above) to the end of the I2SEx_TxRxDMACplt(), outside the DMA_NORMAL mode if statement - my problem was fixed :)

ASELSTM, 2021-04-12:
Hi @lukasnee,

Thank you for this report.  If the issue is systematically reproducible on your side, would you please give us more details on how to reproduce it and if possible share the project so that we can reproduce the same behavior and give you an answer.

With regards,

ASELSTM, 2021-05-18:
Hi @lukasnee,

Any update regarding this issue to move this discussion forward?

With regards,

lukasnee, 2021-06-28:
I don't understand what is unclear. I also found a video where someone else discovered the same missing callback issue: https://youtu.be/lNBrGOk0XzE?t=1383

ASELSTM, 2021-07-02:
Hi @lukasnee,

Thank you for this detailed report. Please allow me to close this thread as answer has been posted within the pull request #15.

With regards,

closed 2021-07-02 by ASELSTM