STMicroelectronics/STM32CubeF3 #17: missing static in stm32f3xx_hal_rcc.c

CanastraRF, 2022-01-03:
Hi
In stm32f3xx_hal_rcc.c are two const array

```
uint8_t const aPLLMULFactorTable[16U] = { 2U,  3U,  4U,  5U,  6U,  7U,  8U,  9U,
                                       10U, 11U, 12U, 13U, 14U, 15U, 16U, 16U};
uint8_t const aPredivFactorTable[16U] = { 1U, 2U,  3U,  4U,  5U,  6U,  7U,  8U,
                                         9U,10U, 11U, 12U, 13U, 14U, 15U, 16U};

```

They are not declared in stm32f3xx_hal_rcc.h.
So they should be static.

Please fix this

RKOUSTM, 2022-01-07:
Hi @CanastraRF,

Thank you for your report. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you again for you report.

With regards,

RKOUSTM, 2022-01-07:
ST Internal Reference: 120490

closed 2023-03-07 by TOUNSTM