STMicroelectronics/STM32CubeF3 #18: Is it possible to add PDM2CPM in F3?

VictorTagayun, 2022-04-07:
Just like what I did in G4, I was able to add manually PDM2PCM [https://github.com/STMicroelectronics/STM32CubeG4/issues/31](https://github.com/STMicroelectronics/STM32CubeG4/issues/31).
Is it also possible to add in F3? I asked in ST community for this problem [https://community.st.com/s/question/0D53W00001U2weXSAR/you-can-also-use-pdm2pcm-on-stm32f3-in-cubemx](https://community.st.com/s/question/0D53W00001U2weXSAR/you-can-also-use-pdm2pcm-on-stm32f3-in-cubemx).

RKOUSTM, 2022-04-11:
Hi @VictorTagayun,

The point you reported has been already treated at [ST Community](https://community.st.com/s/question/0D53W00001U2weXSAR/you-can-also-use-pdm2pcm-on-stm32f3-in-cubemx), please allow me to close this duplicated issue now.

Thank you for your comprehension. We are looking forward to reading from you again.

With regards,

RKOUSTM, 2022-04-11:
Duplicate of [STM32CubeG4](https://github.com/STMicroelectronics/STM32CubeG4/issues/31)

closed 2022-04-11 by RKOUSTM

VictorTagayun, 2022-04-11:
Why you cannot add PDM2PCM in F3?