STMicroelectronics/STM32CubeF3 #21: FLASH_OB_DisableWRP() disables write protection of all pages

P5pDev, 2023-03-16:
Problem in
[https://github.com/STMicroelectronics/STM32CubeF3/blob/master/Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_flash_ex.c](url)

**Describe the set-up**
- STM32F302CCT6 on custom HW
- STM32CubeIDE 1.11.2
- STM32Cube FW_F3 V1.11.3 (but no changes to this file since in latest commit)
- ST-Link on NUCLEO-F072RB with FW V2J40M27

**Describe the bug**
Because of calling `HAL_FLASHEx_OBErase()` https://github.com/STMicroelectronics/STM32CubeF3/blob/e805a235c431db2e01cea47d795fa83d597c64af/Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_flash_ex.c#L679 the `FLASH_OBR` register is reset and because of that the `OB->WRPx` contains 0xffff

This means that the following instructions
https://github.com/STMicroelectronics/STM32CubeF3/blob/e805a235c431db2e01cea47d795fa83d597c64af/Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_flash_ex.c#L687 https://github.com/STMicroelectronics/STM32CubeF3/blob/e805a235c431db2e01cea47d795fa83d597c64af/Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_flash_ex.c#L697 https://github.com/STMicroelectronics/STM32CubeF3/blob/e805a235c431db2e01cea47d795fa83d597c64af/Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_flash_ex.c#L707 https://github.com/STMicroelectronics/STM32CubeF3/blob/e805a235c431db2e01cea47d795fa83d597c64af/Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_flash_ex.c#L717 will always write 0xffff to the registers.

In my opinion, the solution would be 
`OB->WRPx &= WRPx_Data` 
instead, because the `WriteProtectPage `
was already initialized correctly : https://github.com/STMicroelectronics/STM32CubeF3/blob/e805a235c431db2e01cea47d795fa83d597c64af/Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_flash_ex.c#L649

Please compare `FLASH_OB_EnableWRP()`: https://github.com/STMicroelectronics/STM32CubeF3/blob/e805a235c431db2e01cea47d795fa83d597c64af/Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_flash_ex.c#L519

TOUNSTM, 2023-04-11:
ST Internal Reference: 149343

ALABSTM, 2023-08-25:
Fixed in commit 121e4039445ba59f284ebff013d8b3782d503e9f