STMicroelectronics/STM32CubeF3 #23: Double definition of  ADC12_CSR_ADRDY_XXX_XXX

ecrin4102, 2023-10-13:
In the file stm32f303xc.h, 
the constant ADC12_CSR_ADRDY_OVR_SLV_Pos, ADC12_CSR_ADRDY_OVR_SLV_Msk and ADC12_CSR_ADRDY_OVR_SLV  are defined twice.

Moreover, the equivalent constant for ADC34 are not defined.
Line and file reference:
https://github.com/STMicroelectronics/STM32CubeF3/blob/121e4039445ba59f284ebff013d8b3782d503e9f/Drivers/CMSIS/Device/ST/STM32F3xx/Include/stm32f303xc.h#L1999C9-L1999C42
````C
#define ADC34_CSR_ADRDY_EOS_SLV          ADC34_CSR_ADRDY_EOS_SLV_Msk           /*!< End of regular sequence flag of the slave ADC */
#define ADC12_CSR_ADRDY_OVR_SLV_Pos      (20U)                                 
#define ADC12_CSR_ADRDY_OVR_SLV_Msk      (0x1U << ADC12_CSR_ADRDY_OVR_SLV_Pos) /*!< 0x00100000 */
#define ADC12_CSR_ADRDY_OVR_SLV          ADC12_CSR_ADRDY_OVR_SLV_Msk           /*!< Overrun flag of the slave ADC */
#define ADC34_CSR_ADRDY_JEOC_SLV_Pos     (21U)
````

should be replaced by

`````C
#define ADC34_CSR_ADRDY_EOS_SLV          ADC34_CSR_ADRDY_EOS_SLV_Msk           /*!< End of regular sequence flag of the slave ADC */
#define ADC34_CSR_ADRDY_OVR_SLV_Pos      (20U)                                 
#define ADC34_CSR_ADRDY_OVR_SLV_Msk      (0x1U << ADC34_CSR_ADRDY_OVR_SLV_Pos) /*!< 0x00100000 */
#define ADC34_CSR_ADRDY_OVR_SLV          ADC34_CSR_ADRDY_OVR_SLV_Msk           /*!< Overrun flag of the slave ADC */
#define ADC34_CSR_ADRDY_JEOC_SLV_Pos     (21U)
````

RJMSTM, 2023-10-24:
ST Internal Reference: 164694

TOUNSTM, 2024-02-26:
Fixed in 53c8834f42263af4754833980c702ab60c972bab

closed 2024-02-26 by TOUNSTM